space-station-os/space_station_os #5: [Integration] Design reflecting DEMO & Integration scheme

yuyuqq, 2024-11-21:


yuyuqq, 2024-12-15:
Here is a proposal of GitHub repository structure design:
Will be discussed in the rechnical meeting in Dec. 16th 2024.

[image]

Ohara124c41, 2024-12-15:
The added management and maintenance costs are worth the design approach. We will be able to utilize an iterative structure while focusing on the most recent content and archiving/redacting unused content. We just have to make sure we never break the integration modules.

yuyuqq, 2024-12-22:
I have completed moving demo#1 to its own repo. The repo name is "demo_nauka_incident2021_estimate." The content is the same as the original "space_station_gnc" in the "space_station_os" repo. 

Demo creaters can go ahead and start working on creating demos by adding a new repo.

**The demo repository shall have the prefix of ""demo_" so that people can find it is for a demo.**
I will let people know in the next technical meeting and documentation site.

I left the original space_stattion_os repo as it was we could start modifying for the integration.

closed 2024-12-22 by yuyuqq