space-station-os/space_station_os #14: [Demo1] Display space station in Demo1

yuyuqq, 2024-12-04:
Currently Demo1 uses rviz for display. Making it connect with Issac Sim in Linux.

siddarth09, 2025-01-08:
https://github.com/david-dorf/spaceros_gz_demos

can we use the urdf from this? to simulate in both gazebo and issac

dheerubhai-101, 2025-01-08:
Hi, I think it shouldn't be an issue. Just that there is a signficant size difference in the URDFs when I tested them. Apart from that it should be fine

dheerubhai-101, 2025-01-08:
And yeah, I remember these being defined as SDF files, so they may not be compatible with Isaac Sim. We will have to check.

siddarth09, 2025-01-09:
https://discourse.ros.org/t/nvidia-isaac-sim-latest-release-update-with-gazebo-bridge-and-new-features/26130

check this out, i think we can convert easily to USD format for ISSAC sim easily

siddarth09, 2025-08-06:
@dheerubhai-101 Are you still working on this ?

siddarth09, 2025-08-07:
No problem, let us know if you would be interested to work on something else

Closing this issue for now

closed 2025-08-07 by siddarth09