space-station-os/space_station_os #20: [Demo1] Demo for thermal controls

siddarth09, 2025-01-08:


siino14, 2025-01-10:
Siddarth 
Thanks for making new issue. 
If your room mate has github account please assign him for this issue.
and please attach the proposal file. 

Thanks for your hard work !!

siddarth09, 2025-01-10:
He needs to be given access to the project, his username is @Adityamohan260498, if you add him I shall assign him to this project. He also is ready with a demo for the next technical meeting. 

No problem, I would love to be part of this team and the project until it's final flight demo! it would be an honour to work towards it!

siino14, 2025-01-10:
Thank you for your support ! 
I sent invitation to his github account. Please accept the invitation.

yuyuqq, 2025-01-11:
@Adityamohan260498 is added to the task.
@siddarth09 are you still responsible for the task?  Please let me know.

Also, when you create a task, please provide: 
- User story (simple description of what you do, and a feature/functionality from the perspective of the end-users, and why you need this. *)
- Acceptance criteria (providing a clear definition of "done." including conditions of satisfaction)

A sentence is enough for both if it is understandable to the team.

* If you have a proposal document, provide a link to it.

siddarth09, 2025-01-11:
> [@Adityamohan260498](https://github.com/Adityamohan260498) is added to the task. [@siddarth09](https://github.com/siddarth09) are you still responsible for the task? Please let me know.
>
> Also, when [...]

Yes sir, I am helping him out with figuring out the bugs and finishing the code, @Adityamohan260498 Could you please update this with your proposal, and explain how your demo works for thermal controls?

Adityamohan260498, 2025-01-11:
Hello sir,

Thank you for adding me to the task.

I’ve updated the README to include all relevant details about the ATCS simulation. The log output section provides a clear indication of the system’s expected behavior, which aligns with the definition of "done" as requested.

If there’s anything specific you think is missing or needs improvement, please let me know, and I’ll address it promptly.

Link to the repository : [https://github.com/space-station-os/demo_thermal_control](Thermal Control)

siddarth09, 2025-01-18:
Requesting for done, since the demo was shown in the last meeting

closed 2025-01-18 by siddarth09

siino14, 2025-01-18:
Thanks for your hard work!!
Could you share the presentation material in the last technical meeting? (for both of you)

siddarth09, 2025-01-18:
> [presentation-1](https://1drv.ms/p/c/d6ea94a46a13b8d9/EbREEZxrpURFtd2plOwO5KcBZwFB1PGzHS3pnRKI_NjvhQ?e=8EIILO)
> 
> Please find the attached link for the presentation

@siino14 Sir it's mentioned here

Adityamohan260498, 2025-01-22:
[SpacestationOS_Presentation_1.pptx](https://github.com/user-attachments/files/18498052/SpacestationOS_Presentation_1.pptx)

Please find attached ppt, in case you are unable to open it via link