space-station-os/space_station_os #23: [Demo X] Making repository and attach the proposal materials (ECLSS Amateras)

siino14, 2025-01-10:


yuyuqq, 2025-01-11:
@siino14 When you create a task, please provide:

- User story (simple description of what you do, and a feature/functionality from the perspective of the end-users, and why you need this)
- Acceptance criteria (providing a clear definition of "done." including conditions of satisfaction)

A sentence is enough for both if it is understandable to the team.

siino14, 2025-01-13:
**What we do:** Making repository and attach the proposal of demo

**Feature/Functionality from perspective of end-users:** Repository should be added to see new demo

**Why we need this:** To add new demo

**Acceptance criteria:** Making repository and attach the proposal of demo

closed 2025-01-24 by yuyuqq