space-station-os/space_station_os #24: [Demo X] Making repository and attach the proposal materials (ECLSS SSD)

siino14, 2025-01-10:


yuyuqq, 2025-01-11:
@siino14 When you create a task, please provide:

- User story (simple description of what you do, and a feature/functionality from the perspective of the end-users, and why you need this)
- Acceptance criteria (providing a clear definition of "done." including conditions of satisfaction)

A sentence is enough for both if it is understandable to the team.

siino14, 2025-01-13:
@diana1218 
Please provide the following; 
・User story (simple description of what you do, and a feature/functionality from the perspective of the end-users, and why you need this)

・Acceptance criteria (providing a clear definition of "done." including conditions of satisfaction)

Please make repository and attach the proposal materials of demo.

siino14, 2025-01-13:
**What we do:** Making repository and attach the proposal of demo

**Feature/Functionality from perspective of end-users:** Repository should be added to see new demo

**Why we need this:** To add new demo

**Acceptance criteria:** Making repository and attach the proposal of demo

diana1218, 2025-01-14:
Demo written in python have been added. 

Unities: 

1. Hole area: m^2
2. Zvezda module volume:m^3 
3. Oxygen supply: kg
4. Pressure: pascal
5. Oxygen molecular weight: g/mol convert to kg/mol
6. Temperature = 294.3 Kelvin 
7. Oxygen consumption rate: kg/m
8. Escape rate: kg/s convert to kg/day
9. Output time result: Hours and minutes 

The inputs are: Enter the hole area (square meters): 
Enter the angle and direction at which the oxygen flow is escaping (Degrees): 
Enter the number of crew members: 

The outputs are: 
With 3 crew members, the back-up oxygen will last approximately (hrs) hours and (min) minutes.
 Oxygen level/Oxygen Generator Failure
CRITICAL: Oxygen levels have reached zero. Crew members are in danger.