space-station-os/space_station_os #57: [v0.8.3] Implement simulated orbit information

yuyuqq, 2025-04-08:
## User Story:
- Identifying parameteres for orbit design abstraction.
- For implementation of demo_sarj_power_generation, verious orbit information is needed.
- Bringing it to the Integration

## Acceptance Criteria:
- Set abstraction parameters for different orbits
- Simulate orbit dynamics 
- Display it on ground station
- Implement in the Integration
- Prepare for README

k-suga-SD, 2025-04-13:
- Set abstraction parameters for different orbits: by 2025/4/21
- Simulate orbit dynamics
  - Implement time integration method: by 2025/5/5
  - Consider J2 term: by 2025/4/21
  - Consider air drag: by 2025/4/28
  - Consider solar radiation: by 2025/5/5
- Display it on ground station: by 2025/5/19
- Implement in the Integration: by 2025/6/2
- Prepare for README: at any time

k-suga-SD, 2025-08-08:
Until now, I have implemented simulations of attitude and orbital dynamics as demos. Of these, I will integrate the orbital dynamics into the main body of the space station OS (the attitude dynamics have already been implemented in space_station_gnc).

## Policy:
- Use attitude data published in "space_station_gnc/src/physics_motion.cpp"

## My task:
- understand "space_station_gnc/src/thruster_matrix.cpp" because thruster matrix data is required to simulate orbit dynamics
  - Make readme
  - include this file
- understand "space_station_gnc/src/physics_motion.cpp" because attitude dynamics is implemented in this file
  - make readme
  - use topics in it

## unclear points
- Thruster assign matrix
- Thruster force
- thruster uncertainty model
- Mass property of space station (total mass, inertia tensor and center of mass)

k-suga-SD, 2025-08-08:
working branch is https://github.com/k-suga-SD/space_station_os_orbit/tree/v0.8.5-feature/57-orbit

k-suga-SD, 2025-08-10:
I made PR.
I have not done any test (unit/integration). If there are any acceptance criteria for PR, could you tell me that?

k-suga-SD, 2025-08-13:
@siddarth09 
Yes, I will try them.
Is visualization of orbit you mean like this? Now I have implemented test materials (main function, CMakeList.txt, mockup) and done git push. I'm making README.txt to instruct to run test. After that, I will make PR again.

[image]

siddarth09, 2025-08-13:
@k-suga-SD Yes sir, this is something which I wanted so that it's easier for anybody to understand. Also, our GNC nodes are LVLH frames. Are we converting that to ECI, or will you use the LVLH frame later? Additionally, most of the subscribed topics are incorrect. Please check the GNC nodes to have a better understanding. 
I also noticed a string which asks TLE, can you please help me understand what this is, because we might need to publish this topic by GNC. 

Please use ``feature/57-dynamics`` branch to push your commits, I have refactored some ROS parts for parameter usage too.

siddarth09, 2025-08-14:
@k-suga-SD thanks a lot for explaining me the concepts. I understand them now. But the topics subscribed are not the same which we are being used in GNC, please check ``physics_motion.cpp`` for more information regarding this.

k-suga-SD, 2025-08-16:
@siddarth09 I appreciate you refactoring the code.
Are what I should do next those?
- Check `orbit_dynamics.cpp` in the branch `feature/57-dynamics`
- Make PR to merge my test code and visualization code to `feature/57-dynamics`
- Read `physics_motion.cpp` and understand published topics

k-suga-SD, 2025-08-17:
@siddarth09 
orbit_dynamics install was not in CMakeList.txt, so I made a PR to modify that.
And when I did this in `feature/57-dynamics`,
```
$ colcon build --symlink-install
$ source install/setup.sh
$ ros2 run space_station_gnc orbit_dynamics 
```
This error occured.
```space_station_gnc/tests/orbit_dynamics/params.yaml
[ERROR] [1755436084.935177123] [orbit_dynamics_node]: initial.tle_line2 is empty. Set a valid TLE line 2.
[INFO] [1755436084.938017978] [orbit_dynamics_node]: OrbitDynamicsNode ready. dt=0.250 s, pub=0.500 s, RK4=false
orbit_dynamics: /usr/include/eigen3/Eigen/src/Core/Product.h:96: Eigen::Product<Lhs, Rhs, Option>::Product(const Lhs&, const Rhs&) [with _Lhs = Eigen::Matrix<double, 3, -1>; _Rhs = Eigen::Matrix<double, -1, 1>; int Option = 0; Eigen::Product<Lhs, Rhs, Option>::Lhs = Eigen::Matrix<double, 3, -1>; Eigen::Product<Lhs, Rhs, Option>::Rhs = Eigen::Matrix<double, -1, 1>]: Assertion `lhs.cols() == rhs.rows() && "invalid matrix product" && "if you wanted a coeff-wise or a dot product use the respective explicit functions"' failed.
[ros2run]: Aborted```

siddarth09, 2025-08-17:
@k-suga-SD I have merged your PR to the feature branch, now you should be able to make changes to your code and test it out

k-suga-SD, 2025-08-19:
@siddarth09 
I made PR [#134](https://github.com/space-station-os/space_station_os/pull/134). Could you check it?

siddarth09, 2025-08-21:
@k-suga-SD remove your build and install folder. Then build the entire workspace with 

``bash
colcon build --symlink-install 
``

k-suga-SD, 2025-08-21:
@siddarth09 
It still doesn't work. When I run `colcon build --symlink-install`, I get a lot of warnings. Could these be the cause?
Sorry for long console output.

```
sugawara@spica:~/ssos_ws $ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Skip end-of-life distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "iron"
Skip end-of-life distro "jade"
Add distro "jazzy"
Add distro "kilted"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Skip end-of-life distro "melodic"
Skip end-of-life distro "noetic"
Add distro "rolling"
updated cache in /home/sugawara/.ros/rosdep/sources.cache

sugawara@spica:~/ssos_ws $ rosdep install --from-paths src --ignore-src -r -y
#All required rosdeps installed successfully

sugawara@spica:~/ssos_ws $ colcon build --symlink-install
[0.454s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/sugawara/ssos_ws/install/space_station_thermal_control' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.455s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/sugawara/ssos_ws/install/space_station_gnc' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.455s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/sugawara/ssos_ws/install/space_station_eclss' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.455s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/sugawara/ssos_ws/install/space_station_description' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.455s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/sugawara/ssos_ws/install/space_station_control' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.455s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/sugawara/ssos_ws/install/space_station_communication' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.457s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/sugawara/ssos_ws/install/space_station_thermal_control' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.457s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/sugawara/ssos_ws/install/space_station_gnc' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.457s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/sugawara/ssos_ws/install/space_station_eclss' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.457s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/sugawara/ssos_ws/install/space_station_control' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> space_station_eclss
Starting >>> space_station_communication
Starting >>> space_station_control
Starting >>> space_station_description
Starting >>> space_station_gnc
Finished <<< space_station_description [0.86s]
Finished <<< space_station_communication [0.88s]
--- stderr: space_station_control
CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:38 (rosidl_target_interfaces)

CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
  Use rosidl_get_typesupport_target() and target_link_libraries() instead of
  rosidl_target_interfaces()
Call Stack (most recent call first):
  CMakeLists.txt:44 (rosidl_target_interfaces)

---
[... 219 lines cut ...]
[orbit_dynamics_mock-2] [INFO] [1755787930.821564795] [orbit_dynamics_mock]: Published: gnc/t_fwd_sim
[orbit_dynamics_mock-2] [INFO] [1755787930.821637157] [orbit_dynamics_mock]: Published: gnc/attitude_body_to_eci
[orbit_dynamics_mock-2] [INFO] [1755787930.821661433] [orbit_dynamics_mock]: Published: gnc/bias_thruster_cmd
[orbit_dynamics-1] [INFO] [1755787930.821747794] [orbit_dynamics_node]: Subscribed: gnc/t_fwd_sim
[orbit_dynamics_mock-2] [INFO] [1755787931.744168154] [orbit_dynamics_mock]: Published: gnc/t_fwd_sim
[orbit_dynamics_mock-2] [INFO] [1755787931.744307944] [orbit_dynamics_mock]: Published: gnc/attitude_body_to_eci
[orbit_dynamics_mock-2] [INFO] [1755787931.744412060] [orbit_dynamics_mock]: Published: gnc/bias_thruster_cmd
[orbit_dynamics-1] [INFO] [1755787931.744568086] [orbit_dynamics_node]: Subscribed: gnc/t_fwd_sim
[orbit_dynamics_mock-2] [INFO] [1755787932.666463737] [orbit_dynamics_mock]: Published: gnc/t_fwd_sim
[orbit_dynamics_mock-2] [INFO] [1755787932.666537185] [orbit_dynamics_mock]: Published: gnc/attitude_body_to_eci
[... cut ...]
[orbit_dynamics-1] [INFO] [1755787933.588992921] [orbit_dynamics_node]: Subscribed: gnc/t_fwd_sim
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[orbit_dynamics_mock-2] [INFO] [1755787933.821805733] [rclcpp]: signal_handler(signum=2)
[orbit_dynamics-1] [INFO] [1755787933.821993005] [rclcpp]: signal_handler(signum=2)
[INFO] [orbit_dynamics_mock-2]: process has finished cleanly [pid 334791]
[INFO] [orbit_dynamics-1]: process has finished cleanly [pid 334789]

sugawara@spica:~/ssos_ws $ ros2 run space_station_gnc draw_3d_plot.py
No executable found
```

k-suga-SD, 2025-08-22:
@siddarth09 
I tried these command,
```
$ cd ~/ssos_ws
$ rm -rf install/ log/ build/
$ chmod +x src/space_station_os/space_station_gnc/scripts/draw_3d_position.py
$ colcon build --symlink-install
$ sudo rosdep init
$ rosdep update
$ rosdep install --from-paths src --ignore-src -r -y
$ source install/setup.sh
$ ros2 launch space_station_gnc orbit_dynamics.launch.py
$ ros2 run space_station_gnc draw_3d_plot.py
```
But `No executable found` appeared again.
Are my commands correct?

And tree structure is like this.
```
src/space_station_os/space_station_gnc/
├── action
│   └── Unloading.action
├── CMakeLists.txt
├── config
│   ├── config.json
│   ├── params.yaml
│   └── ros_config.yaml
├── include
│   └── space_station_gnc
│       ├── orbit_dynamics.hpp
│       └── thruster_matrix.hpp
├── launch
│   ├── gnc_core.launch.py
│   ├── gnc_rviz.launch.py
│   ├── gnc_rviz_thruster.launch.py
│   └── orbit_dynamics.launch.py
├── LICENSE
├── media
│   ├── 001_start.png
│   ├── 100_emergency_begin.png
│   └── 102_emergency_reaction.png
├── package.xml
├── README.md
├── resource
│   └── space_station_gnc
├── rviz
│   └── ssos.rviz
├── scripts
│   ├── draw_3d_position.py
│   └── jacob_cmgs_casadi.py
├── setup.py
├── src
│   ├── control_torque.cpp
│   ├── demo1a_nauka_incident_estimate.cpp
│   ├── demo1b_crisis_mainengine.cpp
│   ├── demo1c_small_incident.cpp
│   ├── demo1d_sitl.cpp
│   ├── L_p_func.cpp
│   ├── orbit_dynamics.cpp
│   ├── orbit_dynamics_mock.cpp
│   ├── physics_motion.cpp
│   ├── physics_sensor.cpp
│   ├── sense_estimate.cpp
│   ├── space_station_gnc.cpp
│   ├── thruster_matrix.cpp
│   └── torque_collector.cpp
├── tests
│   ├── L_p_func.cpp
│   ├── orbit_dynamics
│   │   ├── rosbag_record.sh
│   │   └── rosbag_to_csv.py
│   ├── pseudoinv_executable
│   ├── test_jacob.cpp
│   └── test_quat.cpp
└── urdf
    ├── ISS_stationaryFromNasa.urdf
    ├── meshes
    └── SD_SpaceStation_Ver05.urdf
```
Is path of `draw_3d_position.py` correct?

siddarth09, 2025-08-22:
@k-suga-SD  I reinstalled, and I was able to run the draw_3d_plot.py. Though all the commands you tried are absolutely correct. Anyway, the previous problem was occurring due to the node not being closed or shut down properly. If you are still facing the issue, remove the entire space_station repo, and clone the feature/57-dynamics branch 

[image]

closed 2025-09-07 by k-suga-SD

k-suga-SD, 2025-09-13:
@siddarth09 @snktshrma @yuyuqq 
In the latest version of the branch "feature/57-dynamics", only gravity effect is considered in orbit dynamics. So I added J2 and air drag effect. I have not made PR yet.

I did those commands, but no message or plot appears after `ros2 run space_station_gnc draw_3d_position.py`. Are there any mistakes?
```
$ cd ~/ssos_ws
$ rm -rf install/ log/ build/
$ chmod +x src/space_station_os/space_station_gnc/scripts/draw_3d_position.py
$ colcon build --symlink-install
$ sudo rosdep init
$ rosdep update
$ rosdep install --from-paths src --ignore-src -r -y
$ source install/setup.sh
$ ros2 launch space_station_gnc orbit_dynamics.launch.py
$ ros2 run space_station_gnc draw_3d_position.py
```

siddarth09, 2025-09-14:
@k-suga-SD 

Let me know if you made a commit, the ``draw_3d_position.py`` is opening in my PC. Let me try changing the code to show a simpler visualisation.

k-suga-SD, 2025-09-22:
@siddarth09 
I'm sorry, but I made a mistake. I ran these commands in one terminal.
```
$ ros2 launch space_station_gnc orbit_dynamics.launch.py
$ ros2 run space_station_gnc draw_3d_position.py
```
To run correctly, two terminals are required. I finally realized that.

siddarth09, 2025-09-24:
@k-suga-SD Let us know if the orbital dynamics are ready for integration into v0.8.6-dev

k-suga-SD, 2025-09-29:
@siddarth09  Now they are ready for integration into v0.8.6-dev.