space-station-os/space_station_os #73: ModuleNotFoundError: No module named 'em' when "colcon build",

k-suga-SD, 2025-06-28:
When run "colcon build" for the main branch, this error occurs.
 
```shell-session
(base) user@hostname:~/workspace/ssos $ mkdir src
(base) user@hostname:~/workspace/ssos $ cd src/
(base) user@hostname:~/workspace/ssos/src $ git clone https://github.com/space-station-os/space_station_os.git
Cloning into 'space_station_os'...
remote: Enumerating objects: 1102, done.
remote: Counting objects: 100% (127/127), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 1102 (delta 99), reused 96 (delta 80), pack-reused 975 (from 1)
Receiving objects: 100% (1102/1102), 91.52 MiB | 7.66 MiB/s, done.
Resolving deltas: 100% (428/428), done.
(base) user@hostname:~/workspace/ssos/src $ ls
space_station_os
(base) user@hostname:~/workspace/ssos/src $ cd ..
(base) user@hostname:~/workspace/ssos/src $ colcon build --symlink-install
Starting >>> ground_station
Starting >>> space_station_base
Starting >>> space_station_communication
Starting >>> space_station_eclss
Starting >>> space_station_electrical
Starting >>> space_station_gnc
Starting >>> space_station_service
Starting >>> space_station_thermal
--- stderr: space_station_eclss
CMake Error at /opt/ros/humble/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:59 (message):
  execute_process(/home/user/software/miniconda3/bin/python3 -m
  rosidl_adapter --package-name space_station_eclss --arguments-file
  /home/user/workspace/ssos/src/build/space_station_eclss/rosidl_adapter__arguments__space_station_eclss.json
  --output-dir
  /home/user/workspace/ssos/src/build/space_station_eclss/rosidl_adapter/space_station_eclss
  --output-file
  /home/user/workspace/ssos/src/build/space_station_eclss/rosidl_adapter/space_station_eclss.idls)
  returned error code 1:

  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
               ^^^^^^
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/main.py", line 53, in main
      abs_idl_file = convert_to_idl(
                     ^^^^^^^^^^^^^^^
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/__init__.py", line 18, in convert_to_idl
      from rosidl_adapter.msg import convert_msg_to_idl
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/msg/__init__.py", line 16, in <module>
      from rosidl_adapter.resource import expand_template
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 19, in <module>
      import em

  ModuleNotFoundError: No module named 'em'

Call Stack (most recent call first):
  /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:44 (rosidl_generate_interfaces)

---
Failed   <<< space_station_eclss [13.5s, exited with code 1]
Aborted  <<< space_station_base [14.4s]
Aborted  <<< ground_station [14.4s]
Aborted  <<< space_station_gnc [14.6s]
Aborted  <<< space_station_electrical [14.7s]
Aborted  <<< space_station_thermal [14.8s]
Aborted  <<< space_station_communication [15.6s]
Aborted  <<< space_station_service [15.9s]

Summary: 0 packages finished [16.1s]
  1 package failed: space_station_eclss
  7 packages aborted: ground_station space_station_base space_station_communication space_station_electrical space_station_gnc space_station_service space_station_thermal
  2 packages had stderr output: space_station_eclss space_station_gnc
 ```

I have already installed "empy". Actually,
 ```shell-session
user@hostname:~/workspace/ssos/src $ python3 -c "import em; print('empy is installed')"
empy is installed
```

yuyuqq, 2025-06-28:
Thank you for reporting this issue.

The error `ModuleNotFoundError: No module named 'em'` during `colcon build` is likely caused by a Python environment mismatch, specifically involving Miniconda.

ROS2's build system (`rosidl_adapter`, `rosidl_cmake`, etc.) expects Python packages such as `empy` (used for message and service generation) to be available in the **system Python environment**, typically `/usr/bin/python3`.

However, in your environment, the build process is using Python from Miniconda:
/home/user/software/miniconda3/bin/python3

This causes the `em` module, even if installed in your user environment, to be unavailable to the ROS 2 toolchain.

## Suggestion to solve:

Please install `empy` into the system Python environment used by ROS2:

```bash
sudo apt update
sudo apt install python3-empy
```
After that, rebuild the workspace:

```bash
cd ~/your_workspace/
colcon build --symlink-install
```
Please let us know if it builds fine.

k-suga-SD, 2025-06-29:
Thank you for your reply.

I did this command.
```console
$ conda deactivate
```
After that, I did those commands you suggested.

```console
user@hostname:~/workspace/ssos/src $ sudo apt update
[sudo] password for user:
Get:1 http://security.ubuntu.com/ubuntu jammy-security InRelease [129 kB]
Get:2 http://packages.ros.org/ros2/ubuntu jammy InRelease [4,682 B]
Err:2 http://packages.ros.org/ros2/ubuntu jammy InRelease
  The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
Hit:3 http://archive.ubuntu.com/ubuntu jammy InRelease
Hit:4 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:5 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Fetched 134 kB in 1s (102 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
273 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.ros.org/ros2/ubuntu jammy InRelease: The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
W: Failed to fetch http://packages.ros.org/ros2/ubuntu/dists/jammy/InRelease  The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
user@hostname:~/workspace/ssos/src $ sudo apt install python3-empy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-empy is already the newest version (3.3.4-2).
0 upgraded, 0 newly installed, 0 to remove and 273 not upgraded.
user@hostname:~/workspace/ssos/src $ colcon build --symlink-install
Starting >>> ground_station
Starting >>> space_station_base
Starting >>> space_station_communication
Starting >>> space_station_eclss
Starting >>> space_station_electrical
Starting >>> space_station_gnc
Starting >>> space_station_service
Starting >>> space_station_thermal
Finished <<< ground_station [0.17s]
Finished <<< space_station_communication [0.17s]
Finished <<< space_station_electrical [0.18s]
Finished <<< space_station_service [0.18s]
Finished <<< space_station_thermal [0.19s]
Finished <<< space_station_base [0.23s]
--- stderr: space_station_eclss
CMake Error at /opt/ros/humble/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:59 (message):
  execute_process(/home/user/software/miniconda3/bin/python3 -m
  rosidl_adapter --package-name space_station_eclss --arguments-file
  /home/user/workspace/ssos/src/build/space_station_eclss/rosidl_adapter__arguments__space_station_eclss.json
  --output-dir
  /home/user/workspace/ssos/src/build/space_station_eclss/rosidl_adapter/space_station_eclss
  --output-file
  /home/user/workspace/ssos/src/build/space_station_eclss/rosidl_adapter/space_station_eclss.idls)
  returned error code 1:

  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
               ^^^^^^
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/main.py", line 53, in main
      abs_idl_file = convert_to_idl(
                     ^^^^^^^^^^^^^^^
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/__init__.py", line 18, in convert_to_idl
      from rosidl_adapter.msg import convert_msg_to_idl
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/msg/__init__.py", line 16, in <module>
      from rosidl_adapter.resource import expand_template
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 19, in <module>
      import em

  ModuleNotFoundError: No module named 'em'

Call Stack (most recent call first):
  /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:44 (rosidl_generate_interfaces)

---
Failed   <<< space_station_eclss [3.13s, exited with code 1]
Aborted  <<< space_station_gnc [3.19s]

Summary: 6 packages finished [3.58s]
  1 package failed: space_station_eclss
  1 package aborted: space_station_gnc
  2 packages had stderr output: space_station_eclss space_station_gnc
```

It seems that python3-empy is already the newest version.
And there is still the line "execute_process(/home/user/software/miniconda3/bin/python3 -m".
So,
```console
$ python
Command 'python' not found, did you mean:
  command 'python3' from deb python3
  command 'python' from deb python-is-python3

$ which python3
/usr/bin/python3
```
So I ran this command.
```console
$ alias python='/usr/bin/python3'
$ python
Python 3.10.12 (main, May 27 2025, 17:12:29) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>exit()
```

But the same error occurred again.

siddarth09, 2025-07-02:
Pleas let me know if you are able to resolve this issue with above instructions

k-suga-SD, 2025-07-08:
@siddarth09
Thank you so much for your detailed instructions!
> ROS 2 (colcon and ament) internally caches the Python interpreter path at first build time, from the active environment.

I didn't know that.
Step 1 and 2 worked, but when I tried Step 3,
```console
ModuleNotFoundError: No module named 'casadi'
```
appeared. So I installed 'casadi' additionally by
```console
$ pip install casadi
```
After that, new errors appears.
```console
user@hostname:~/workspace/ssos/src $ colcon build --symlink-install
Starting >>> ground_station
Starting >>> space_station_base
Starting >>> space_station_communication
Starting >>> space_station_eclss
Starting >>> space_station_electrical
Starting >>> space_station_gnc
Starting >>> space_station_service
Starting >>> space_station_thermal
Finished <<< space_station_communication [0.14s]
Finished <<< space_station_thermal [0.15s]
Finished <<< space_station_base [0.19s]
Finished <<< space_station_service [0.19s]
Finished <<< ground_station [0.21s]
Finished <<< space_station_electrical [15.7s]
[Processing: space_station_eclss, space_station_gnc]
--- stderr: space_station_eclss
In file included from /home/user/workspace/ssos/src/space_station_os/space_station_eclss/src/ors_systems/src/deionization_chamber.cpp:1:
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ors_system/deionization_bed.h: In constructor ‘IonizationBed::IonizationBed()’:
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ors_system/deionization_bed.h:33:10: warning: ‘IonizationBed::is_active’ will be initialized after [-Wreorder]
   33 |     bool is_active;
      |          ^~~~~~~~~
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ors_system/deionization_bed.h:29:12: warning:   ‘double IonizationBed::water_’ [-Wreorder]
   29 |     double water_;
      |            ^~~~~~
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/src/ors_systems/src/deionization_chamber.cpp:6:1warning:   when initialized here [-Wreorder]
    6 | IonizationBed::IonizationBed()
      | ^~~~~~~~~~~~~
In file included from /home/user/workspace/ssos/src/space_station_os/space_station_eclss/src/ars_systems/adsorbent_tank_1.cpp:1:
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ars_system/adsorbent_tank.hpp: In constructor ‘AdsorbentBed1::AdsorbentBed1()’:
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ars_system/adsorbent_tank.hpp:37:8: warning: ‘AdsorbentBed1::is_active_’ will be initialized after [-Wreorder]
   37 |   bool is_active_;
      |        ^~~~~~~~~~
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ars_system/adsorbent_tank.hpp:34:42: warning:   ‘double AdsorbentBed1::co2_buffer_’ [-Wreorder]
   34 |   double co2_, moisture_, contaminants_, co2_buffer_;
      |                                          ^~~~~~~~~~~
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/src/ars_systems/adsorbent_tank_1.cpp:7:1: warning:   when initialized here [-Wreorder]
    7 | AdsorbentBed1::AdsorbentBed1()
      | ^~~~~~~~~~~~~
In file included from /home/user/workspace/ssos/src/space_station_os/space_station_eclss/src/ars_systems/adsorbent_tank_1.cpp:1:
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ars_system/adsorbent_tank.hpp:36:10: warning: ‘AdsorbentBed1::previous_error_’ will be initialized after [-Wreorder]
   36 |   double previous_error_ = 0.0;
      |          ^~~~~~~~~~~~~~~
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ars_system/adsorbent_tank.hpp:35:10: warning:   ‘double AdsorbentBed1::temperature_’ [-Wreorder]
   35 |   double temperature_ = 300.0;
      |          ^~~~~~~~~~~~
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/src/ars_systems/adsorbent_tank_1.cpp:7:1: warning:   when initialized here [-Wreorder]
    7 | AdsorbentBed1::AdsorbentBed1()
      | ^~~~~~~~~~~~~
In file included from /home/user/workspace/ssos/src/space_station_os/space_station_eclss/src/ars_systems/adsorbent_tank_2.cpp:1:
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ars_system/adsorbent_tank.hpp: In constructor ‘AdsorbentBed2::AdsorbentBed2()’:
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ars_system/adsorbent_tank.hpp:65:8: warning: ‘AdsorbentBed2::is_active_’ will be initialized after [-Wreorder]
   65 |   bool is_active_;
      |        ^~~~~~~~~~
/home/user/workspace/ssos/src/space_station_os/space_station_eclss/include/space_station_eclss/ars_system/adsorbent_tank.hpp:63:10: warning:   ‘double AdsorbentBed2::co2_buffer_’ [-Wreorder]
[... 61 lines cut ...]
gmake[2]: *** [CMakeFiles/desiccant2.dir/build.make:76: CMakeFiles/desiccant2.dir/src/ars_systems/desiccant_tank_2.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:684: CMakeFiles/desiccant2.dir/all] Error 2
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/electrolysis.dir/build.make:76: CMakeFiles/electrolysis.dir/src/ors_systems/src/electrolysis.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:852: CMakeFiles/electrolysis.dir/all] Error 2
c++: fatal error: Killed signal terminated program cc1plus
[... cut ...]
gmake[2]: *** [CMakeFiles/clean_water_tank.dir/build.make:76: CMakeFiles/clean_water_tank.dir/src/wrs_systems/src/water_processor_assembly/product_water_tank.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1048: CMakeFiles/clean_water_tank.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< space_station_eclss [1min 11s, exited with code 2]
Aborted  <<< space_station_gnc [1min 18s]

Summary: 6 packages finished [1min 19s]
  1 package failed: space_station_eclss
  1 package aborted: space_station_gnc
  2 packages had stderr output: space_station_eclss space_station_gnc
```
I am also debugging in parallel.

siddarth09, 2025-07-08:
Are you running multiple builds at the same time ? and do you have enough space on your PC?

siddarth09, 2025-07-08:
https://github.com/space-station-os/space_station_os/tree/v0.8.4

Follow this readme, but remember while cloning, just add ```-b v0.8.4```  to get the correct branch. I have uninstalled and reinstalled multiple times, and I am not able to reproduce this error. Request you to do this

k-suga-SD, 2025-07-09:
I'm sorry but I cloned the main branch. 
I tried to clone v0.8.4 branch, but failed. Do I need any authorization?

```console
user@hostname:~/workspace/ssos/src $ git clone --recurse-submodules https://github.com/space-station-os/space_station_os.git -b v0.8.4
Cloning into 'space_station_os'...
remote: Enumerating objects: 1126, done.
remote: Counting objects: 100% (151/151), done.
remote: Compressing objects: 100% (63/63), done.
remote: Total 1126 (delta 113), reused 112 (delta 86), pack-reused 975 (from 1)
Receiving objects: 100% (1126/1126), 91.52 MiB | 4.03 MiB/s, done.
Resolving deltas: 100% (442/442), done.
Submodule 'space_station_eclss' (git@github.com:space-station-os/space_station_eclss.git) registered for path 'space_station_eclss'
Submodule 'space_station_gazebo' (git@github.com:space-station-os/space_station_gazebo.git) registered for path 'space_station_gazebo'
Submodule 'space_station_gnc' (git@github.com:space-station-os/space_station_gnc.git) registered for path 'space_station_gnc'
Submodule 'space_station_mission_control' (git@github.com:space-station-os/space_station_mission_control.git) registered for path 'space_station_mission_control'
Cloning into '/home/user/workspace/ssos/src/space_station_os/space_station_eclss'...
The authenticity of host 'github.com (20.27.177.113)' can't be established.
ED25519 key fingerprint is SHA256:*******************************************
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'github.com' (ED25519) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:space-station-os/space_station_eclss.git' into submodule path '/home/user/workspace/ssos/src/space_station_os/space_station_eclss' failed
Failed to clone 'space_station_eclss'. Retry scheduled
Cloning into '/home/user/workspace/ssos/src/space_station_os/space_station_gazebo'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:space-station-os/space_station_gazebo.git' into submodule path '/home/user/workspace/ssos/src/space_station_os/space_station_gazebo' failed
Failed to clone 'space_station_gazebo'. Retry scheduled
Cloning into '/home/user/workspace/ssos/src/space_station_os/space_station_gnc'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:space-station-os/space_station_gnc.git' into submodule path '/home/user/workspace/ssos/src/space_station_os/space_station_gnc' failed
Failed to clone 'space_station_gnc'. Retry scheduled
Cloning into '/home/user/workspace/ssos/src/space_station_os/space_station_mission_control'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:space-station-os/space_station_mission_control.git' into submodule path '/home/user/workspace/ssos/src/space_station_os/space_station_mission_control' failed
Failed to clone 'space_station_mission_control'. Retry scheduled
Cloning into '/home/user/workspace/ssos/src/space_station_os/space_station_eclss'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:space-station-os/space_station_eclss.git' into submodule path '/home/user/workspace/ssos/src/space_station_os/space_station_eclss' failed
Failed to clone 'space_station_eclss' a second time, aborting
```
And I haven't installed Docker, so I will install it later.

siddarth09, 2025-07-09:
> I tried to clone v0.8.4 branch, but failed. Do I need any authorisation?

You are already part of the organisation, I guess you have not added SSH keys, please follow the tutorial so that you can easily install it. Once your SSH is set up on GitHub, you can follow the required steps as mentioned in the README. 

https://docs.github.com/en/authentication/connecting-to-github-with-ssh

This documentation should help you. 

Since it's not pushed to the main branch, I have added the SSH link, but when we are integrating to the main branch, I will update this to https link

2. To install Docker, you can https://docs.docker.com/engine/install/ubuntu/ use this. After you do this, follow the step which tells you to build the Docker file ( again, this step will not be required when we push it to the main branch).

k-suga-SD, 2025-07-12:
Thank you, I have never made SSH key for GitHub.
I'm following the tutorial, finding additional information and struggling. But I feel like it will go well.

siddarth09, 2025-07-12:
Let me know how i can assist you, if you want to I can join a call and help you out too!