space-station-os/space_station_os #93: Wiki page creation

yuyuqq, 2025-07-16:


siddarth09, 2025-07-24:
@yuyuqq  @snktshrma need your help for GNC theory and implementation wiki

yuyuqq, 2025-10-04:
@PavanSandaka 
Would you please suggest a list of contents you would like to know about GNC as a new comer.

In my mind I would write these three items:
1. Use cases
2. How to use it (manual)
3. Theory behind it

I would be great if you could make the items detail (especially 3).

siddarth09, 2025-10-08:
@yuyuqq @PavanSandaka 

https://github.com/space-station-os/space_station_os/wiki/GNC

please check the documentation and let me know if you feel some information is missing, please feel free to make changes to the wiki if you feel its too lengthy too. 

I have also created PR #93

closed 2025-10-30 by siddarth09