space-station-os/space_station_os #117: [v0.8.5] [Enhancement] Improvements in Thermal control logic

siddarth09, 2025-08-01:
Current Thermal control nodes have random temperature and thermal properties associated with that particular node. The improvement will enhance the package with the help of a plugin or ways to add thermal properties individually for the respective nodes. 
@4BlakeThompson will be setting up a GUI to visualise the thermal network using QT5 or Kivy. 

Other contributors are welcomed to enhance this package to make it more realistic

4BlakeThompson, 2025-08-02:
@siddarth09  How should the layout be generated for the network visualization? Right now there is just one base link and then nodes connected to it, but I'm assuming it will become much more complicated and it won't be as easy as just branching from the base link in a circle. What kind of graph data will it have to be able to handle long term? Do we want spatial information, grouping, or hierarchy to influence how the network is laid out? Or based on relationships like a force-directed layout? Also for coolant information, should that just be off to the side or will it become part of the network somehow?
Thanks in advance for your help.

4BlakeThompson, 2025-08-03:
I will come up with a way of representing them, but could you tell me in what ways they will get more complicated? That way I can design it with dynamically handling the future changes in mind. For example will there always be a base link? Will there be different types of nodes? Other ways it will change from the current version?

commit 295c9a26 (2025-08-12): Refactor: YAML-driven architecture for thermal nodes

- Removed URDF dependency from thermal solver node.
- Implemented YAML-driven configuration for handling multiple thermal nodes (supports 50+ nodes).
- Verified integration with ECLSS: service calls work as expected, and thermal network behavior remains consistent with v0.8.4.
- Improved maintainability and modularity by externalizing parameters.

Closes #117, #113.
closed 2025-08-13 by siddarth09