space-station-os/space_station_os #118: [v0.8.5 or v0.8.6] [Enhancement] OPEN-MCT improvements with additonal custom plugins

siddarth09, 2025-08-01:
User: 

Open-MCT provides basic plugins required for a mission task, but contributors can add more plugins based on the requirements for the ground control station. 

Acceptance Criteria: 
1. The Plugin should be able to handle data received by ROS2 (present), CCSDS (after v0.8.7) 
2. The developer/ contributor can create custom plugins according to the requirements. 

@yuyuqq: Need help regarding the requirements for subsystem visualisation of telemetry status in mission control.

yuyuqq, 2025-08-03:
@siddarth09 
What are the differences from Issue #94?
If there is significant overlap, please consider merging the two issues.

siddarth09, 2025-08-03:
The Issue #94 is to add existing open mct plugins to our bare bones version, this issue to enhance the ground station with more custom plugins, and change in backend data.